Componolit/gneiss #3: Fix SYNC implementation for Genode

senier, 2019-02-27:
In the Genode platform implementation we use `Block::Packet_descriptor::END` to signal a SYNC request. Not sure whether this is right - check and remove if this does not denote a sync operation.

If we make `SYNC` a no-op on Genode, are we going to deliver a `SYNC` event with `ERROR` set? Or a syncable flag for the device?

jklmnn, 2019-02-27:
I think we have to rethink the semantic of sync in general. Currently we're supplying a packet with sync (including block numbers). Is block information something we want to have in sync? Is syncing single blocks even supported (and if so, is it always) or useful?

senier, 2019-03-06:
The semantics of `SYNC` is to write all requests to persistent storage before processing any new requests. This is a non-blocking operation (subsequent requests are queued or rejected).

commit 567bf90c (2019-03-06): implement empty sync for genode
fixes #3
commit 48451af2 (2019-03-15): implement empty sync for genode
fixes #3
closed 2019-03-15 by senier