Componolit/gneiss #6: Add `Trim` operation

senier, 2019-03-06:
The `TRIM` operation is beneficial for optimizing SSD storage. For `Trim` the request type is equivalent to `Read`/`Write`. The request is:

```
procedure Trim (D : in out Component.Block_Device; R : in out Request);
```

jklmnn, 2019-03-29:
Are trim request required to be acknowledged? Since they only tell the block device that it can discard a range of blocks. Since the server is not required to provide anything to the client it shouldn't be necessary to acknowledge the trim. Aside from this the sync shouldn't be needed to acknowledged either (as it is only a marker for the server and the client can enqueue further requests after the sync and be sure that anything before the sync will be executed).
If we have to acknowledge sync and trim request the "empty" implementation on Genode would become a bit more complicated since we have to insert fake acknowledges which will also not trigger the event handler.

senier, 2019-04-01:
`Trim` is just a hint to the underlying component, i don't think it has to be acknowledged.

Regarding `Sync`: This operation actually takes offset and block range (cf. 3.51- SYNCHRONIZE CACHE (10) of [Direct Access Block commands](https://www.seagate.com/files/staticfiles/support/docs/manual/Interface%20manuals/100293068j.pdf)). We should make it a "normal" request again to support those scenarios.

Regarding missing platform support: When a primitive is unsupported on a platform in general, we should not fake it, but give the client a mechanism to detect it and cope with it. Idea: Add a `supported()` predicate (similar to `ready()`) which is a precondition to `Enqueue()` and is implemented by the platform. Before enqueuing a request, the component has to check whether its supported and take measures if not.

commit 623f84e8 (2019-03-29): add trim request w/o ack
fixes #6
commit 7e8946eb (2019-04-01): add trim request w/o ack
fixes #6
closed 2019-04-01 by senier