Componolit/gneiss #10: Generic session interface

jklmnn, 2019-03-08:


jklmnn, 2019-03-08:
Servers are a bit more special as they require a step of indirection more. To support multiple sessions of each session type each type requires a "meta"-session that handles requests. The meta session triggers an event when a session request arrives (which obviously can't happen in the session that is not yet created). The meta session is instantiated as a conventional session and in its event handler it takes requests and instantiates per-connection sessions of the actual requested type.

senier, 2019-03-10:
I was just wondering about the server session: If we define a server session interface using a generic package spec, how could a component use multiple implementations of *the same* server interface? Say you want to use two different block drivers in your component - both implementations realize the same body of the generic interface and are thus mutually exclusive in the same component.

Idea: Server sessions are implemented in a separate package which the server session generic is instantiated with. The body of the generic server session then just calls (renames?) operations of that package.

@jklmnn: Does that sound feasible? Other ideas?

jklmnn, 2019-03-10:
The generic interface package is not implemented by the server implementation but can be instantiated multiple times with different implementations. With the above example this would look like:

```Ada
type State is private;

procedure Event1 (S : State);
procedure Event2 (S : State);

package Session1 is new Session (State, Event1);
package Session2 is new Session (State, Event2);

procedure Handle_Request1 (S : State; C : in out System.Address);
procedure Handle_Request2 (S : State; C : in out System.Address);

package Server1 is new Server (State, Session1, Handle_Request1);
package Server2 is new Server (State, Session2, Handle_Request2);
```

In this case two completely different server types could be used. Although the `Handle_Request` procedure probably needs to check for certain properties of the incoming request to decide which implementation is responsible. I'm not sure how Genode would handle announcing the same session interface multiple times. We should check how those cases would be handled.

About your idea, I'm not sure if I understand it correctly. We should talk about this tomorrow.

senier, 2019-03-15:
Closed by #11

closed 2019-03-15 by senier