Componolit/gneiss #15: Support multiple LOG instances

senier, 2019-03-16:
The new client log interface assumes that we have one single connection to a log server that is already initialized when we start using it. We would not be able to implement, e.g. a log splitter with the current interface.

Our LOG session should get a `Create` function and an `Initialize` procedure as the block session. In `Initialize` we could then configure e.g. different log backends or target facilities.

jklmnn, 2019-03-16:
How would such an interface be implemented on Genode? I'm not sure if there is a specific LOG session client interface. But you're right a log splitter would be desirable.

senier, 2019-03-16:
Sure, there [is one](https://github.com/genodelabs/genode/blob/master/repos/base/include/log_session/client.h). However, for now I'd suggest we use the shortcut version and implement `Create` and `Initialize` as no-ops. Unless using the client session is a no-brainer. If I remember correctly, we tried to use the LOG client interface in the past and had some trouble there.

Of cause, we'll also need a finalize operation.

Another thing that the interface IMHO is lacking is some `Sync` (or `Flush`) method to drain the log output. When connecting to our block interface, this would trigger a SYNC request after writing the log message. What do you think?

jklmnn, 2019-03-16:
Implementing using the log client you linked doesn't seem too hard (at least after my experiences with the Block interface, we don't need callbacks or a generic package here) so I'd implement this directly without creating intermediate empty functions.

While I second that there are flush/sync operations in other log implementations we should take a short look on their semantics on different systems to not make a too block session specific log sync. Also even if our interface supports calling flush/sync (I'd personally prefer flush in this context) Genode doesn't seem to. I cannot see anything that implements this, so our flushs would never arrive at the log server.

senier, 2019-03-17:
> Implementing using the log client you linked doesn't seem too hard (at least after my experiences with the Block interface, we don't need callbacks or a generic package here) so I'd implement this directly without creating intermediate empty functions.

Cool!

> While I second that there are flush/sync operations in other log implementations we should take a short look on their semantics on different systems to not make a too block session specific log [...]

I'm not aware of any platform that has a flush for its log interfaces. Still, I think it should be there.

commit b33220fd (2019-03-18): log client session
fixes #15
closed 2019-03-18 by senier