Componolit/gneiss #18: Create interface for read-only configuration

senier, 2019-03-19:
Interface to retrieve a component-specific configuration which is a string of arbitrary format. The platform implementation has a well-known location for the configuration (Genode: ROM named "config", Muen: memory region named "config" (or similar), POSIX: environment variable named "CAI_CONFIG" ointing to file).

The config interface is realize as generic package with a procedure with a single in-parameter as formal parameter:

```Ada
generic
   with procedure Parse (Data : in String);
package CAI.Configuration.Client
   function Create return CAI.Configuration.Client_Session;
   procedure Initialize (C : in out CAI.Configuration.Client_Session);
   function Initialized (C : in CAI.Configuration.Client_Session) return Boolean;
   procedure Load (C : in out CAI.Configuration.Client_Session);
   procedure Finalize (C : in out CAI.Configuration.Client_Session);
   ...
end CAI.Configuration.Client;
```

jklmnn, 2019-04-25:
Should we call Parse automatically if the configuration has been updated or should the application decide?

jklmnn, 2019-04-25:
Also should there be a guarantee for `Parse` to be called if `Load` is called?

senier, 2019-04-25:
I'd suggest `Parse` is guaranteed to be called on `Load` and it is automatically called on configuration updates.

Btw, how do we handle configuration updates? On Genode I guess a signal is registered for an update of the underlying ROM? How do we do that on POSIX? [FAM](https://en.wikipedia.org/wiki/File_Alteration_Monitor) comes to mind. Or `inotify`, but that's Linux specific.

jklmnn, 2019-04-26:
While FAM looks like the only relatively platform independent solution it doesn't seem to be something we want to depend on. The official homepage is from 2003 and only available via the web archive and the version currently available dates back to at least 2005 (excluding Debian maintainer updates).

I would opt for inotify. The block implementation is also linux specific and in case we want to support another posix platform I would just modify/add the platform specific parts to the code.

senier, 2019-04-26:
I got the same impression regarding FAM and would prefer inotify, too. We should consider naming the platform `linux` then, as it's really not POSIX (or platform independent).

jklmnn, 2019-04-26:
I moved the platform naming discussion into #51 since the issue came up multiple times now and we should discuss this more intensively.

Since inotify uses blocking reads on special file descriptors I couldn't integrate this approach easily into our signal dispatching. I implemented a file descriptor registration which enables each interface to register file descriptor based signal handlers and callbacks. The main loop is now handled by a select call on all registered file descriptors. Signalling support has been achieved by using [signalfd](http://man7.org/linux/man-pages/man2/signalfd.2.html) which provides signals via file descriptors. Although it is not possible for interfaces to register their signals as they're still handled centrally via a single file descriptor.

senier, 2019-04-29:
The configuration test fails on Linux with SIGBUS sometimes. Also, not all changes are detected (especially when changing the file with vim).

Most likely, the problem is that we `mmap` the file only on initialize. When the file gets deleted, the mapped file becomes invalid. We need to change the code such that `mmap` is done every time `Load` is executed and `munmap` is done after calling `Parse`. If `mmap` fails, we pass an empty buffer to `Parse`.

jklmnn, 2019-05-02:
Fixed by #56.

closed 2019-05-02 by jklmnn