Componolit/gneiss #32: Implement timer interface

senier, 2019-04-04:
This interface allows users to request the current monotonic time (similar to `Clock` in `Ada.Real_Time`). We should also think about function to retrieve precision and bounds of the timer.

jklmnn, 2019-04-17:
Proposed timer client interface:
```Ada

package Cai.Timer is

   type Time is new Duration;

   function "+" (Left : Time; Right : Duration) return Time;
   function "-" (Left : Time; Right : Duration) return Time;
   function "-" (Left : Time; Right : Time) return Duration;
   
   type Client_Session is limited private;

end Cai.Timer
```
```Ada
with Cai.Types;

package Cai.Timer.Client is

   function Create return Client_Session;

   function Initialized (C : Client_Session) return Boolean;

   procedure Initialize (C : in out Client_Session; Cap : Cai.Types.Capability);

   function Clock (C : Client_Session) return Time with
      Volatile_Function;

   procedure Finalize (C : in out Client);

end Cai.Timer.Client;
```

jklmnn, 2019-04-29:
Fixed by #54.

closed 2019-04-29 by jklmnn