Componolit/gneiss #37: Implement NBD server

senier, 2019-04-08:
The NBD server implements a block server session and exports if over a socket.

jklmnn, 2019-04-26:
I did some searches and NBD seems to be the only reasonable way to implement a block device driver in user space on Linux without a custom kernel module.
There's also a simple [abstraction layer](https://github.com/acozzette/BUSE) which we probably could partially reuse.

senier, 2019-04-26:
I couldn't find any user-space alternative to NBD for implementing block devices on Linux either. Linking our components with more untrusted code like BUSE feels like a step into the wrong direction, though. Especially as this would only be applicable to Linux hosts.

If we pursue the NBD idea, we should rather define a network (stream) server interface for CAI and implement the NBD protocol using RecordFlux. This way we'd gain a trusted component that can run on different platforms and we would extend CAI with another useful session type that is tested in a real-world setting.

An interesting different direction is the one we discussed lately, moving our components into the Linux kernel by implementing a Linux kernel platform in the runtime and CAI. Running [Ada code inside the kernel](https://github.com/alkhimey/Ada_Kernel_Module_Framework) has been done. Implementing block servers should be trivial then.

jklmnn, 2019-04-26:
I didn't consider BUSE as a dependency to link against but rather a proof of concept that we could look into when creating our own implementation. I also opened #51 to discuss the platform definition issue separately.

closed 2019-10-28 by senier