Componolit/gneiss #43: Build Block.Client.Job interface

senier, 2019-04-12:
Implement a state machine approach on top of our current (low-level) block client interface.

jklmnn, 2019-04-12:
The job interface is meant to track the state of an operation on the block device on a higher level than requests. A job is an operation on a consecutive number of blocks that can be `Read`, `Write`, `Sync` or `Trim`. The size, amount and order of the requests is hidden in the implementation and the state of the job will be hold by the component all the time. It will have the following states: `Raw`, `Pending`, `Ok`, `Error`.
Furthermore all interfaces are designed to let the reader of any data choose how much it wants to read. A theoretical spec would look like:

```Ada
generic
   with procedure Write (Jid        :     Job_Id;
                         Block_Size :     Size;
                         Data       : out Buffer;
                         Length     :     Count);
package Cai.Block.Client.Jobs

   type Job_Status is (Raw, Pending, Ok, Error);

   type Job_Id is private;
   type Job is private;

   procedure Initialize (J         : out Job;
                         Start     :     Id;
                         Length    :     Count;
                         Operation :     Request_Kind);

   procedure Update (C : in out Client_Session;
                     J : in out Job);

   function Status (J : Job) return Job_Status;

   function Get_Id (J : Job) return Job_Id;

   procedure Read (J      : in out Job;
                   C      : in out Client_Session;
                   Data   :    out Data;
                   Length : in out Count);

   procedure Release (J : in out Job;
                      C : in out Client_Session);

end Cai.Block.Client.Jobs;
```

SInce the job resides in the package state an identifier is required for the `Write` procedure to identify the job that is pending without aliasing the original object. A regular workflow would look as follows:
 - write
   - Initialize, `Raw -> Pending`
   - Update, `Pending -> Pending|Ok|Error`
     - Write, fill the out parameter with data
   - Release, `Ok|Error -> Raw`
 - read
   - Initialize, `Raw -> Pending`
   - Update, `Pending -> Pending|Ok|Error`
   - Read, `Ok -> Ok`, the out parameter gets filled
   - Release, `Ok|Error -> Raw`

For both read and write the caller provides the maximal length of the buffer in block size times blocks with the `Length` parameter and the callee changes it to block size times number of blocks written to the buffer. This implies that `Length` cannot grow in this procedure but only stay constant or shrink.

jklmnn, 2019-04-15:
To prevent recursions all procedures that are called from within `Run` (basically all generic parameters) must not call `Run` by themselves.

closed 2019-10-28 by senier