Componolit/gneiss #50: Dispatcher session capability

jklmnn, 2019-04-23:
The procedures `Session_Request`, `Session_Accept`, `Session_Cleanup` must only be called in the `Dispatch` procedure of the dispatcher. This is currently not enforced. To enforce this the `Dispatch` procedure will receive a limited private in paramter that is required to be passed to each of this three methods. As it is a limited private type it cannot be copied or stored outside the `Dispatch` procedure.

senier, 2019-07-16:
The block dispatcher provides a session label with the `Session_Request` procedure. This label can be used to identify the requesting client. It is then passed to the `Session_Accept` procedure. Inbetween it could have been changed. Since `Session_Accept` uses the label to identify the actual session to connect the label shouldn't be changed as any other session is probably not available or cannot be safely connected to. This information should be instead passed via the dispatcher capability

jklmnn, 2019-07-24:
Do we want to make the label still available to the component implementing the dispatcher? It could decide which connections it accepts via the label. This functionality could also be implemented in the initialize procedure which still gets the label passed. Not initializing would have the same effect as not accepting the session (see #72).

commit d2aad142 (2019-06-26): muen dispatcher capability
fixes #50
closed 2019-08-26 by jklmnn