Componolit/gneiss #52: Make image functions of log interface generic

senier, 2019-04-29:
Currently we need a public type for the image functions of the log interface to cast own types. A more elegant solution is a generic function with a formal discrete type.

senier, 2019-07-16:
Idea: Make the image function generic in its input type, add a `base` parameter with supported bases betwee 2..16. For commonly used types in the `Interfaces` package we provide instances.

jklmnn, 2019-07-23:
Maybe we should split this functionality from the log package completely. It was initially thought as a convenience function collection for the log package. But I often use it independently. Also I reimplemented different parser functions for different Integer types/Duration. Maybe we should move all these functions together into a separate package (whatever that would be called, I'm open for suggestions).

senier, 2019-07-24:
I second that idea. As for the name: `Cai.Representation`? Alternatively, `Cai.Image` and different functions.

jklmnn, 2019-07-25:
In the runtime we have this kind of functionality in the `Strings` package. I'd suggest this as it is shorter and still on topic.

senier, 2019-07-25:
Sounds good.

closed 2019-08-26 by jklmnn