Componolit/gneiss #67: Timer event support

senier, 2019-07-01:
Implement a session that triggers an event after a specific amount of time.

- events are relative (in X seconds)
- data type is our 'time' data type (which equals the standard Duration type)
- One single oneshot timer is supported
- Timers cannot be canceled

jklmnn, 2019-07-01:
The timer interface will be changed to
```Ada
generic
   with procedure Event;
package Cai.Timer.Client with
   SPARK_Mode
is

   --  Returns new session object
   --
   --  @return  Timer client session object
   function Create return Client_Session;

   --  Checks if session is initialized
   --
   --  @param C  Timer client session object
   --  @return   True if session is initialized else False
   function Initialized (C : Client_Session) return Boolean;

   --  Initialized timer session
   --
   --  @param C    Timer client session object
   --  @param Cap  System capability
   procedure Initialize (C : in out Client_Session; Cap : Cai.Types.Capability) with
      Pre => not Initialized (C);

   --  Returns a monotonic clock value
   --
   --  This function returns a monotonic rising clock value for each call.
   --  The returned value does not necessarily start at 0.
   --
   --  @param C  Timer client session object
   --  @return   Current clock value
   function Clock (C : Client_Session) return Time with
      Volatile_Function,
      Pre => Initialized (C);

   procedure Set_Timeout (C : in out Client_Session; T : Duration) with
      Pre => Initialized (C);

   --  Finalizes timer session
   --
   --  @param C  Timer client session object
   procedure Finalize (C : in out Client_Session) with
      Pre  => Initialized (C),
      Post => not Initialized (C);

end Cai.Timer.Client;
```

`Timer` will become a generic package that takes an `Event` procedure which is called once a timeout is triggered. The timeout can be set via `Set_Timeout`. The time will be counted from the call of this procedure and the `Event` procedure will be called once. The timer precision is platform dependent and the only guarantee available is that AT LEAST the time given in the timeout has passed.

jklmnn, 2019-07-03:
Fixed by #68.

closed 2019-07-03 by jklmnn