Componolit/gneiss #97: Run multiple components in one "process"

senier, 2019-09-25:
This could also mean that all components run on a microcontroller with a bare-metal runtime.

jklmnn, 2020-03-03:
https://iotality.com/armcm-access-levels/
https://iotality.com/armcm-svc/

jklmnn, 2020-03-09:
About the fancy variant of different "programs" using the same runtime there's an interesting question on StackOverflow: https://stackoverflow.com/questions/25172834/how-to-create-static-linked-shared-libraries

jklmnn, 2020-06-05:
After some research and discussions I've come to the conclusion that creating a binary with multiple similar static libraries requires a significant amount of work. Since all applications have the same namespace and symbol names (at least for Components) we would have to link and locate the libraries manually as the usual tools would always find collisions. This would also include a patched gnatbind to correctly generate the elaboration code without reinitializing the runtime.

An alternative solution would be to move each application into its own namespace. As a convention the initialization could still be done in `<Name>.Component` (now it is just `Component`). This would make it possible to include all applications into the main project file and have them linked correctly. It would still allow to keep our current dynamic linking procedure. If the toplevel package is named equally to the library the symbol could be generated from the library name.
On Genode the package name doesn't matter at all since we export a specific symbol name to C++.
cc @Fabien-Chouteau

jklmnn, 2020-08-28:
After some separate discussion, using a separate namespace for each component and compiling them into their own static libraries seems to be the best solution. The initial idea was to build all components as static libraries and register them via their elaboration code. However since the elaboration code is not executed in units that are not withed some code has to be generated anyway. So instead of registering components there will be a generated startup procedure that calls all component constructors in order.

Fabien-Chouteau, 2020-08-28:
What do you mean by using separate namespace? Prefixing all package names?

In which situation do you want to elaborate units that are not withed?

jklmnn, 2020-08-28:
> What do you mean by using separate namespace? Prefixing all package names?

Packages that implement a component all have to be a child package of a package named after the component since they are all compiled into a single binary.

> In which situation do you want to elaborate units that are not withed?

The original idea was that all components are compiled into static libraries that are withed into the project. When their elaboration code is run they register on a globally known instance. However since their code is never used directly their elaboration code is never run.