Componolit/gneiss #107: Gneiss init on Linux

jklmnn, 2019-10-28:


jklmnn, 2019-10-28:
To run more complex scenarios on Linux with multiple components a configuration and orchestration of the components is required.The idea is a custom Init component written in C that loads the configuration, sets up the channels, forks itself and then starts the components.

The configuration is defined in XML (unless there's any valuable reason for anything else). As short draft it could look as follows:

```XML
<config>
    <component name="hello_world">
        <file name="hello_world.so"/>
        <service label="log" server="printf_log"/>
    </component>

    <component name="printf_log">
        <file name="printf_log.so"/>
        <service label="hello_world_log" client="hello_world"/>
    </component>
</config>
```

The `component` node defines a component that is going to be started. The `file` node specifies the binary that implements this component. The `service` node specifies a packet/session/etc. interface where the label is used to identify the channel within the application. The client specifies a `server` attribute that denotes to which server it connects and the server has a `client` attribute that tells it which client is connected to that label.

The components are compiled as shared objects so they can be loaded at runtime. Init is a executable that reads the configuration and sets up the required channels (by distributing the right file descriptors). It then forks for each component into a loader. The loader removes all remaining parts from init that are not required to run its designated component and then loads the shared object.

The shared object component provides the `Construct` procedure either as a symbol or a constructor (depending on which one works better). The main loop of a component is a select over all file descriptors that are registered as callbacks. To register a callback the component is given a function pointer through the capability and that can be used to register a callback.

senier, 2019-10-28:
Looks good so far. The `<service>` node will need some kind of type information to enable init to deduce the kind of file descriptor it needs to distribute.

jklmnn, 2019-11-05:
For each component an information object about the available services and their labels is generated. To make this easier the services are not embedded into the components but as separate nodes:
```XML
<service type="Message" client="client_component" server="server_component" client_label="client_label" server_label="server_label"/>
```

jklmnn, 2019-11-05:
Also I think it is more useful to set the file via an xml attribute and not via a separate node, changing
```XML
<component name="name">
    <file name="filename"/>
</component>
```
to
```XML
<component name="name" file="filename" />
```

jklmnn, 2019-11-05:
Since some resources do not map well onto the server/client model we'll model them with participants. Each participant may access the resource reading (`rd`), writing (`wr`) or both (`rw`).  A resource has a number of maximal (exact?) users of each type:
```XML
<resource type="Message" name="channel1" rw="2" rd="0" wr="0"/>
<resource type="Memory" name="shm1" rw="0" rd="1" wr="1"/>
<component name="controller" file="controller.so">
    <resource type="Message" name="channel1" label="client" mode="rw"/>
    <resource type="Memory" name="shm1" label="client" mode="wr"/>
</component>
<component name="client" file="client.so">
    <resource type="Message" name="channel1" label="controller" mode="rw"/>
    <resource type="Memory" name="shm1" label="controller" mode="rd"/>
</component>
```

jklmnn, 2019-11-13:
Fixed by #117.

closed 2019-11-13 by jklmnn