Componolit/gneiss #108: Packet interface

jklmnn, 2019-10-28:


jklmnn, 2019-10-28:
The packet interface for Gneiss is a generic packet based communication interface. It consists of a message interface (#109) for meta data and control flow and a shared memory (#110) for payload data. The shared memory is splitted into two parts, one for each communication participant.

In the first iteration the packets are only sent via the message interface with larger messages (e.g. 1500b). In a second iteration they will be sent via the shared memory channel. Finally the idea is to send small messages (e.g. <16b) inline together with the meta data and allocate everything larger in the shared memory.

jklmnn, 2020-05-13:
This proposal only contains the session specific interfaces. Session generic interfaces such as initialization are omitted.
```Ada
generic
   type Size is range <>;
   type Byte is (<>);
   type Index is range <>;
   type Buffer is array (Index range <>) of Byte;
package Gneiss.Packet with
   SPARK_Mode
is

   type Descriptor is limited private;

   function Size (D : Descriptor) return Size;

   function Allocated (D : Descriptor) return Boolean;

   function Writable (D : Descriptor) return Boolean;

end Gneiss.Packet;

generic
   type Context is limited private;
   with procedure Event;
   with procedure Generic_Update (C   : in out Client_Session;
                                  D   :        Descriptor
                                  B   :    out Buffer;
                                  Ctx : in out Context);
   with procedure Generic_Read (C   : in out Client_Session;
                                D   :        Descriptor;
                                B   :        Buffer;
                                Ctx : in out Context);
package Gneiss.Packet.Client with
   SPARK_Mode
is

   procedure Allocate (C : in out Client_Session;
                       D : in out Descriptor) with
      Pre  => not Allocated (D),
      Post => (if Allocated (D) then Writable (D));

   procedure Send (C : in out Client_Session;
                   D : in out Descriptor) with
      Pre  => Allocated (D) and then Writable (D),
      Post => not Allocated (D);

   procedure Update (C   : in out Client_Session;
                     D   : in out Descriptor;
                     Ctx : in out Context) with
      Pre  => Allocated (D) and then Writable (D),
      Post => Allocated (D) and then Writable (D);

   procedure Read (C    : in out Client_Session;
                   D    : in out Descriptor;
                   Ctx : in out Context) with
      Pre  => Allocated (D) and then not Writable (D),
      Post => Allocated (D) and then not Writable (D);

   procedure Receive (C : in out Client_Session;
                      D : in out Descriptor) with
      Pre  => not Allocated (D),
      Post => (if Allocated (D) then not Writable (D));

   procedure Free (C : in out Client_Session;
                   D : in out Descriptor) with
      Post => not Allocated (D);

end Gneiss.Packet.Client;
```

Since the client and server semantics are symmetric only the client interfaces are designed here. Each packet has a descriptor that is bound in the application. When sending a packet its descriptor has to be allocated by `Allocate`. If the allocation was successful the descriptor is writable and the allocated memory can be filled by calling `Update` When it is initialized it can be sent with `Send`. This will also free the descriptor (and its memory) again.
To receive a packet an unallocated descriptor is passed to `Receive`. If it is allocated after a packet has been received, otherwise there was no data available. A received packet is never writable. Its contents can be read by calling `Read`. When it has been processed it can be freed with `Free`.

senier, 2020-05-14:
LGTM.

Could there be situations where we want to write a received packet? E.g. inplace modification and forwarding of a packet.

jklmnn, 2020-05-14:
I was thinking about this. However, modifying and sending a received packet would require that sender and receiver can access the same memory. If we use `malloc` on Linux this would be universally possible. If we use a shared memory region to allocate packets we could only send the packet back on the interface that received it but not forward it to another interface (otherwise it would be some kind of magic zero copy between two different shared memory regions). The safe (and platform independent way) is not allowing received packets to be sent again by anyone. For a modification (e.g. a proxy) it could be useful to have a move operation that moves the packet (and the attached memory) between interfaces. This move could then be zero or single copy depending on the platform requirements.

senier, 2020-05-14:
> The safe (and platform independent way) is not allowing received packets to be sent again by anyone.

Yes, let's go for the more general approach and not allow received packets to be writable then.

> For a modification (e.g. a proxy) it could be useful to have a move operation that moves the packet (and the attached memory) between interfaces. This move could then be zero or single copy depending on the platform requirements.

That sounds like a good solution!

jklmnn, 2020-05-14:
> That sounds like a good solution!

I would however implement this solution later and not in the first step (unless we really need it). When we build it a separate discussion might be needed.

senier, 2020-05-14:
> > That sounds like a good solution!
> 
> I would however implement this solution later and not in the first step (unless we really need it). When we build it a separate discussion might be needed.

No, we don't need it right now and I didn't mean to add it in this first version.

jklmnn, 2020-05-20:
Fixed by #177.

jklmnn, 2020-06-17:
Fixed by #186.

jklmnn, 2020-06-24:
Fixed again by #188.

closed 2020-06-24 by jklmnn