Componolit/gneiss #126: Parent protocol and extended init

jklmnn, 2019-11-22:
A simple init that sets up the resources before starting the components is insufficient for more dynamic scenarios. Since allowing direct connects between components does not fulfill the security requirements (as clients could forge labels and no third party would be able to check the connections) a trusted third party is required.

As a solution we should introduce a parent protocol and a parent component that checks policies. This would be Linux specific for the short term. The idea is that the `init` or loader only starts the components (including the parent component) and sets up sockets for the parent protocol.
If a component wants to connect to a server it will send a request to the parent, which will then check if the label and the source of the component is valid. It will then connect to the server directly, and if the server accepts it will send the file descriptor back to the client component. If the server rejects the connection it will reject the connection to the client also.

senier, 2019-11-22:
Does the parent even need to check the label? Couldn't it just use the label associated with the client FD from the configuration?

jklmnn, 2019-11-22:
But the client sends a label to which server it wants to connect. E.g. if there exist two servers the client has to specify which server it wants to connect to which would do it by choosing a label. The parent then has to check if there exists a route to a server and that the route has a matching label attached.
On the other hand if we do the routing completely through the parent we could indeed also pass the label through to the server and only use it for configuration purposes.

senier, 2019-11-25:
I think you should draft a brief design of the new broker component in this ticket, how it interacts with init, which configuration and file descriptors it receives etc. before starting to implement this.

jklmnn, 2019-11-25:
Maybe we should adapt the concept and configuration to the new more dynamic semantics. We basically need two things:
 - A component identifier that is known to the parent and that allows to make routing decisions
 - An connection identifier that is known by the server and that allows to make decisions about dispatching and how to handle which incoming connection (e.g. who to give which partition in case of an partitioner)

# Routing

The routing must be unambiguous. This means that there must be exactly one possible server for each connection attempt by a client. This problem can be parted into three situations:
 - One server is available. This is the easiest part, no matter how many connections of this type are opened by an arbitrary number of clients, they all are routed to this single server.
 - Multiple servers are available and each client opens one connection. In this case for each client it has to be specified to which server it shall connect.
 - Multiple servers are available and each client opens multiple connections (to possibly different servers). Now there needs to be a server for each connection. The problem here is that connections need to be identified to let the client specify which connection will go to which server (without letting the client know the actual servers). The easiest way would be to enable a client to specify connection labels that are available in the config (and maybe require a default server that all unlabeled connections are routed to).

The result is in principle what Genode currently does:
```XML
<config>
  <component name="hello_world" file="libcomponent_hello_world.so">
    <service name="Log" server="logserver"/>
    <service name="Log" label="xml" server="xmllogserver"/>
  </component>
  <component name="logserver"...>
  ...
  </component>
  <component name="xmllogserver"...>
  ...
  </component>
</config>
```

# Identifying connections

Identifying incoming connections can be again split into two smaller problems:
 - Each client opens a single connection. In this case the identification only needs to distinguish between clients. The easiest way would be to pass the client name or to assign a label to each client.
 - A client opens multiple connections. This case is more complicated. Not only might the client open a connection with a label (that could be passed) but it could also open additional multiple connections without labels. Currently we only pass the label of the connection if it has one or an empty string if it doesn't but no information about the client. On Genode this information is available.

A possible solution would be to append the label defined by the client application to the client name (which is basically also label) and to pass it to the server which would then decide on this information how to handle the incoming connection.

The security properties are that a client cannot forge a connection as it were another client. This is made sure due to the fact that the client cannot specify its own name (which is done by the config/parent). While it can freely specify the label of the connection it could only "forge" its own labels that it knows anyway.
The handling of the labels in the server would be done by the component specific configuration in the server.

This is very close to Genodes current design but I cannot see a better solution currently. On Genode the syntax is basically `component_label -> connection_label`. The difference is that there is no real distinction between component and connection as components and therefore also labels can be recursive.
We have to decide for an internal representation and how much this representation should be accessible from the component.
An idea would be to use `:` as simple label separator (and therefore disallow it in label names). Also currently there is only one label string passed to the server component. Either this will be done in the future, too, requiring the component to parse the label components each time. The other options would move the parsing into Gneiss. One would be to pass two separate strings, limiting the flexibility to two labels for each connections or passing an array of strings limiting the maximum length of labels (and possibly wasting memory for shorter labels in the array).

jklmnn, 2019-11-25:
As for the parent protocol itself each component will receive a file descriptor that it can use to make requests and receive file descriptors.

# Client

The client will send a lookup message
```
Lookup:<Type>:<Label>
```
for a specific service type and a label (which can be empty) to the parent. Then the parent will check the policy if the client is allowed to do this request. If this is not the case it will reject the request with `Reject`.
If the client is allowed to connect to the server the parent will check if it knows a server with that type and name in its policy. If this is the case the parent will ask the server if it can serve the request with `Connect:<Component_Label>:<Label>`. If this is the case the server will answer with an `Accept` and a file descriptor. If not, the server will reject. The servers answer will be relayed to the asking client.

# Server

Servers are not required to announce their services as routing is done in a way that clients need an explicit permission to use a service. If a server provides a service that is not intended clients cannot connect to it due to the policy. Also client cannot circumvent the policy as the only way to initially interact with the server is the parent file descriptor that is controlled solely by the parent.
When the server receives a request as described in the Client section it can decide whether it accepts it or not. If it wants to accept it it can open a file descriptor to itself and then send this file descriptor to the parent together with the accept message. If it cannot open a further file descriptor or if it can't handle the request it will reject it.

The idea behind this concept is that no connection can be made without the parent being involved. If each component runs as its own user id even the parent can drop privileges as it is not required to open other components files. Yet this also makes it impossible for the parent to start further components.

senier, 2019-11-25:
> * One server is available. This is the easiest part, no matter how many connections of this type are opened by an arbitrary number of clients, they all are routed to this single server.
>
> * Multiple [...]

I wonder whether this distinction is necessary - can't we just use `server+connection` in all cases?

jklmnn, 2019-11-25:
I made this distinction solely to understand the problem better. If we're supporting the last (and most complex) case we're also automatically supporting the first two.

senier, 2019-11-25:
> The client will send a lookup message
> 
> ```
> Lookup:<Type>:<Label>
> ```
> 
> for a specific service type and a label (which can be empty) to the parent. Then the parent will check the policy if the client is allowed to do this request. If this is not the case it will reject the request with `Reject`.

I think the protocol makes sense. IMHO we should define a binary protocol using RecordFlux, mainly because we can and we avoid writing and proving parsers manually. The same is true for the labels - why not use a binary format instead of test separators.

senier, 2019-11-25:
NB: is the name parent protocol appropriate? Its rather a ... connection broker?

senier, 2019-11-25:
> ```
> <config>
>   <component name="hello_world" file="libcomponent_hello_world.so">
>     <service name="Log" server="logserver"/>
>     <service name="Log" label="xml" server="xmllogserver"/> [...]

I'm not sure I understand this example. The first `<service>` entry is used when the client does not provide a label and the second when it provides the label `xml`?

How would we specify the routing for more complex labels. Say, a client uses "tcp:10.182.4.64" as a label. The label will be handled by the server, but how do we define the routing? Maybe we need two distinct labels, one for routing and one for the server?

jklmnn, 2019-11-25:
> I think the protocol makes sense. IMHO we should define a binary protocol using RecordFlux, mainly because we can and we avoid writing and proving parsers manually. The same is true for the labels - why not use a binary format instead of test separators.

If we use a binary protocol I'd propose a simple TLV protocol that solves the problem with a specific separator. In this case I think we also could encode the message and session types into numbers, e.g. `Lookup:Log:<label>` would become `00 01 00 01 00 05 Label` (2 byte message type, 2 byte session type, 2 byte length, label string).

> NB: is the name parent protocol appropriate? Its rather a ... connection broker?

Parent protocol doesn't seem to be the right name, indeed. Also since the system isn't exactly hierarchical. The "parent" component is rather the connection broker and the interface would be a broker session (although I'm not completely satisfied with the name).

> I'm not sure I understand this example. The first `<service>` entry is used when the client does not provide a label and the second when it provides the label `xml`?

Yes.

> How would we specify the routing for more complex labels. Say, a client uses "tcp:10.182.4.64" as a label. The label will be handled by the server, but how do we define the routing? Maybe we need two distinct labels, one for routing and one for the server?

We shouldn't overuse the labels for dynamic data, it was more thought of some static selector for what can be configured. In this case I'd rather set the label to `tcp1` and configure the server that `tcp1` should connect to `10.182.4.64`. If we encode too much functionality into the label system the meta data we have to transfer outside of the actual session will increase and the functionality will be more difficult to implement on other system. E.g. how would you implement this label on Muen?

senier, 2019-11-25:
> If we use a binary protocol I'd propose a simple TLV protocol that solves the problem with a specific separator. In this case I think we also could encode the message and session types into numbers, [...]

Yes, something like that. Please draft a RecordFlux spec for the protocol and the label format. I'm sure @treiher is happy to help you regarding good and bad protocol design choices.

> Parent protocol doesn't seem to be the right name, indeed. Also since the system isn't exactly hierarchical. The "parent" component is rather the connection broker and the interface would be a broker session (although I'm not completely satisfied with the name).

I can't think of anything more suitable than *broker* right know. The meaning IMHO is on spot.

> We shouldn't overuse the labels for dynamic data, it was more thought of some static selector for what can be configured. In this case I'd rather set the label to `tcp1` and configure the server [...]

Fair enough. I was ignoring static setups. So the labels are really just for selecting a peer, but connection parameters would be negotiated once the client holds the connection. Makes sense.

jklmnn, 2020-01-02:
The current implementation seems to work conceptually. There seems to be only one problem left. https://github.com/Componolit/gneiss/commit/12a77ad8c7441798bf4ce853153678f9c3f88b10 added debug messages to the message client and server components which seem to prevent the problem.
When the problem occurs it seems as both server and client are the same binary. Both send a request for the message service (which only the client should do) with one of them getting denied of course. The original client seems to behave correctly (as it is just doing nothing in absence of an answer) and also the broker doesn't seem to be broken (there are two messages sent and received, and the message received from the "wrong" client is rejected). Only the server behaves weird, although already on initialization (probably before the first request is received).
This doesn't appear anymore when the debug message is printed. But it might also be gone after removing the output and compiling again.

jklmnn, 2020-01-03:
Sometimes the broker starts both components and then nothing happens. I was unsure what exactly happened but it could be that this is the opposite case of what I mentioned above. It might be that in this case both components are started as servers which then do nothing and listen for incoming connections.

jklmnn, 2020-01-03:
The problem seemed to originate from the build system. Gprbuild seems to reuse some objects it should recompile instead. This happened since I was building multiple components into the same directories that had the same units. Depending on the order of the compilation of single objects and the order of the targets it either worked, or it used the wrong component.adb.
I fixed this in https://github.com/Componolit/gneiss/commit/2f70344f78b69bbc7c64ff1138047f5c74017208 by relocating the build tree for each component. The problem is that relocating the build tree also leads to a full recompilation of all libraries. A solution might be to only redefine the Object_Dir of the component. Although these issues should be discussed in #130.

jklmnn, 2020-01-06:
Fixed by #131.

commit fa399dd3 (2019-12-11): disable debug output

fixes #126 #119
closed 2020-01-06 by jklmnn