Componolit/gneiss #146: Set up/document programming infrastructure for PINETIME

senier, 2020-02-20:
Use know-good image for testing the setup.

jklmnn, 2020-03-03:
https://dev.to/aaronc81/flashing-your-pinetime-using-an-st-link-and-openocd-54dd
https://wiki.pine64.org/index.php/PineTime
https://medium.com/@ly.lee/coding-nrf52-with-rust-and-apache-mynewt-on-visual-studio-code-9521bcba6004
https://medium.com/@ly.lee/openocd-on-raspberry-pi-better-with-swd-on-spi-7dea9caeb590

jklmnn, 2020-03-05:
https://forum.pine64.org/showthread.php?tid=8631

jklmnn, 2020-04-02:
Create binary file from ELF:
 - `arm-eabi-objectcopy -O binary main main.bin`

GPR Configuration:
 - `for Connection_Tool use "";`
 - `for Connection_Config_File use "";`

Programming:
 - `program main.bin 0x0 reset`