Componolit/gneiss #156: Memory session on Linux

jklmnn, 2020-03-18:
In the memory session the client requests the size of the shared memory region. On Linux this is a datum that is not transmitted in the request.

senier, 2020-03-18:
I don't understand the implications of the above statement.

jklmnn, 2020-03-19:
The problem ist, that the client selects the desired size of the shared memory. This information has to be passed to both the broker and the server but the current protocol between component and broker does not allow transmitting meta data.

jklmnn, 2020-03-19:
After some discussion a separate meta data field is not required for this session. The client will open the file descriptor with `memfd_create` and use its own resources. It will set a size for this fd and send it to the broker. The broker guarantees to keep the fds size and to seal it. If it cannot apply the required seals (`F_SEAL_SEAL`, `F_SEAL_SHRINK`, `F_SEAL_GROW`) the request will be rejected.

jklmnn, 2020-04-01:
Fixed by #158.

closed 2020-04-01 by jklmnn