Componolit/gneiss #165: Custom parameters for generic operations

senier, 2020-03-30:
My specific issue is with the memory server session, but I guess the issue applies to other sessions, too. Take the following memory server:

```Ada
package Memory_Server is new Memory.Server (Modify, Initialize, Finalize, Ready); 
```

To use the `Modify` operation, we call the `Modify` procedure of the memory server with the session as an argument:

```Ada
Memory_Server.Modify (Servers (Idx).Mem); 
```

The problem now is that `Modify` may need to perform different things depending on the context in which `Memory_Server.Modify` is called (e.g. copy data from shared memory to a buffer or vice versa). Right now, we need to define some package variable, set it before calling `Modify` and performing a different operation depending on that variable.

It would be easier if `Modify` had an additional variable with a generic type.

jklmnn, 2020-04-08:
I noticed that this approach is problematic for the block interface. It defines the `Request` type in `Gneiss.Block.Client` which depends on the `Context` type. Since the requests are often part of the application state the `Context` type also depends on the `Request` type creating a circular dependency. A solution could be changing the block interface accordingly, that all types containing relevant state should be defined in the session package and not inside its children. I would omit the block session for now since we do not have it on Linux and therefore cannot prove it.

jklmnn, 2020-04-09:
`Server.Finalize` and `Server.Initialize` are only called by `Dispatcher.Session_Cleanup` and `Dispatcher.Session_Initialize`. They also need a context argument.

On Linux currently `Finalize` and the server event are both called by the same callback. And both get the `Server_Session` as argument. If `Finalize` is to be called by `Session_Cleanup` which only can be called from within `Dispatch`, the event needs to pass both the server and the dispatcher session. To avoid aliasing there should be two events, one for an IO event and one for an error event. The first will call the server event and the latter will call dispatch. Both get their own session parameter. Also both will get the file descriptor that called them passed.

jklmnn, 2020-05-19:
Fixed by #173.

closed 2020-05-19 by jklmnn